JingwenWang95/DSP-SLAM #1: Segmentation fault when run test on kitti

LongruiDong, 2021-12-02:
Hi:
Thans for contributing such a great project!
I have build DSP-SLAM successfully, but when I run dsp_slam on your kitti 07,  i got  Segmentation fault,
below is terminal output in gdb debugging
```
(gdb) run
Starting program: /home/dlr/Project/DSP-SLAM/dsp_slam Vocabulary/ORBvoc.bin configs/KITTI04-12.yaml /media/dsp-slam/data/kitti/07 map/kitti/07
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

DSP-SLAM: Object Oriented SLAM with Deep Shape Priors.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

[debug]: py::module::import(sys)

[New Thread 0x7ffec9cde700 (LWP 8631)]
....
[New Thread 0x7ffd6f288700 (LWP 8704)]
[debug]: py::module::import(reconstruct.utils)

[debug]: pyCfgPath: configs/config_kitti.json

[New Thread 0x7ffd5f9a7700 (LWP 8711)]
...
[New Thread 0x7ffd52ffd700 (LWP 8732)]
[debug]: io_utils.attr(get_decoder)(pyCfg)

[debug]: strSequencePath: /media/dsp-slam/data/kitti/07

[pydebug] to get_seq	
[pydebug] to Class KITIISequence	
[pydebug] load_calib() done	
[Thread 0x7ffd6f288700 (LWP 8704) exited]
...
[Thread 0x7ffec9cde700 (LWP 8631) exited]
[pydebug] data_type == KITTI	

Thread 1 "dsp_slam" received signal SIGSEGV, Segmentation fault.
0x00007fffd4a16e6c in geos::io::WKBReader::readGeometry() () from /usr/lib/x86_64-linux-gnu/libgeos-3.6.2.so
(gdb) bt
#0  0x00007fffd4a16e6c in geos::io::WKBReader::readGeometry() () at /usr/lib/x86_64-linux-gnu/libgeos-3.6.2.so
#1  0x00007ffe52595e14 in  () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/Shapely.libs/libgeos_c-74dec7a7.so.1.14.2
#2  0x00007ffe525980ee in GEOSGeomFromWKB_buf_r () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/Shapely.libs/libgeos_c-74dec7a7.so.1.14.2
#3  0x00007fffe82d7dae in ffi_call_unix64 () at /usr/lib/x86_64-linux-gnu/libffi.so.6
#4  0x00007fffe82d771f in ffi_call () at /usr/lib/x86_64-linux-gnu/libffi.so.6
#5  0x00007fffb4baa7b4 in _call_function_pointer (argcount=3, resmem=0x7fffffff24a0, restype=<optimized out>, atypes=<optimized out>, avalues=0x7fffffff2480, pProc=0x7ffe525980a0 <GEOSGeomFromWKB_buf_r>, flags=<optimized out>) at /usr/local/src/conda/python-3.7.12/Modules/_ctypes/callproc.c:816
#6  0x00007fffb4baa7b4 in _ctypes_callproc (pProc=0x7ffe525980a0 <GEOSGeomFromWKB_buf_r>, argtuple=0x7ffe52849550, flags=<optimized out>, argtypes=0x7ffe5280d410, restype=<optimized out>, checker=0x0) at /usr/local/src/conda/python-3.7.12/Modules/_ctypes/callproc.c:1188
#7  0x00007fffb4bab02c in PyCFuncPtr_call (self=<optimized out>, inargs=<optimized out>, kwds=0x0) at /usr/local/src/conda/python-3.7.12/Modules/_ctypes/_ctypes.c:4025
#8  0x00007ffff583ac04 in PyObject_Call () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#9  0x00007ffff56f3805 in partial_call () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#10 0x00007ffff583add9 in _PyObject_FastCallKeywords () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#11 0x00007ffff56b13c3 in call_function.lto_priv () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#12 0x00007ffff56b7c77 in _PyEval_EvalFrameDefault () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#13 0x00007ffff56badea in function_code_fastcall () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#14 0x00007ffff56b1416 in call_function.lto_priv () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#15 0x00007ffff56b5d92 in _PyEval_EvalFrameDefault () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#16 0x00007ffff576ee80 in _PyEval_EvalCodeWithName () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#17 0x00007ffff576e29f in PyEval_EvalCodeEx () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#18 0x00007ffff576f93c in PyEval_EvalCode () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#19 0x00007ffff5773b0e in builtin_exec () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#20 0x00007ffff5839f9c in _PyMethodDef_RawFastCallDict () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#21 0x00007ffff583a9d9 in _PyCFunction_FastCallDict () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#22 0x00007ffff56b95f1 in _PyEval_EvalFrameDefault () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#23 0x00007ffff576ee80 in _PyEval_EvalCodeWithName () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#24 0x00007ffff583a0ea in _PyFunction_FastCallKeywords () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#25 0x00007ffff56b1416 in call_function.lto_priv () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#26 0x00007ffff56b7c77 in _PyEval_EvalFrameDefault () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#27 0x00007ffff56badea in function_code_fastcall () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#28 0x00007ffff56b1416 in call_function.lto_priv () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#29 0x00007ffff56b5f46 in _PyEval_EvalFrameDefault () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#30 0x00007ffff56badea in function_code_fastcall () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#31 0x00007ffff56b1416 in call_function.lto_priv () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
#32 0x00007ffff56b5d92 in _PyEval_EvalFrameDefault () at /home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/libpython3.7m.so.1.0
---Type <return> to continue, or q <return> to quit---
...
```

according to my print infos. I noticed the code stuck at 
https://github.com/JingwenWang95/DSP-SLAM/blob/cfce877fb27ac8c9c9cc4ba4e0832a4d109595ad/reconstruct/__init__.py#L5

I am wondering how to solve this , can you give some insights?
Thanks in advance!

JingwenWang95, 2021-12-02:
Hi LongruiDong,

Thanks for reaching out! The segfault seems quite wierd. From where it stucks it seems that it cannot find the module .detector. And I saw that you are running DSP-SLAM from your home directory, my hypothesis is that as you are not running the program under the prgram directory, it might be the case that the Python interpreter cannot locate that module. Have you tried cd into the DSP-SLAM directory and run the program with relative paths? You can simply follow the command in readme.

Also, the Python reconstruction script reconstruct_frame.py is very useful for you to debug. Can you successfully run that script?

Feel free to contact me if you have any questions.

Best
Jingwen

LongruiDong, 2021-12-02:
> Hi LongruiDong,
>
> Thanks for reaching out! The segfault seems quite wierd. From where it stucks it seems that it cannot find the module .detector. And I saw that you are running DSP-SLAM from your [...]

Actually I am running the program in DSP-SLAM directory, the gdb show the full path of each args of raw command.

yes I can also run  reconstruction script reconstruct_frame.py fluently, which may indicates the python can find  detector3d module. 
Furthermore, I change to run DSP-SLAM with offline detector as ReadMe.txt, which also normally get succeed. and I found the map is okay...

So now the online detect mode is the question...

JingwenWang95, 2021-12-02:
Hi! Sorry, I thought you were running it from your home because I saw that you used the absolute path of the executable. Why don't you just use `./dsp-slam`? So the problem now is you can run online mode on Redwood and Freiburg sequences but get errors when running KITTI? It shouldn't happen because if you run it under the DSP-SLAM directory it will automatically include the [current directory](https://github.com/JingwenWang95/DSP-SLAM/blob/master/src/System.cc#L93) to Python path, especially given that you said you can run `reconstruct_frame.py` smoothly...

I have tested from my side and cannot reproduce the error. Did you make sure that everything is the same when you run on KITTI and mono sequences? I'm sorry but I have no idea what might be happening. You can reach out to me via Jingwen.Wang.17@ucl.ac.uk and we can have a more detailed discussion via email.

LongruiDong, 2021-12-03:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Hi, Jingwen:
        I always use ./dsp_slam to run all data set under DSP-SLAM directory.below is terminal output when run reconstruct_frame.py :```(dsp-slam) ***@***.***:~/Project/DSP-SLAM$ python reconstruct_frame.py --config configs/config_kitti.json --sequence_dir /media/dsp-slam/data/kitti/07 --frame_id 100[pydebug] load_calib() done	[pydebug] data_type == KITTI	WARNING - 2021-12-03 16:00:00,092 - collision - No FCL -- collision checking will not work[pydebug] import get_detector3d	/home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/mmdet/core/anchor/builder.py:16: UserWarning: ``build_anchor_generator`` would be deprecated soon, please use ``build_prior_generator``   '``build_anchor_generator`` would be deprecated soon, please use 'load checkpoint from local path: /media/dsp-slam/weights/maskrcnn/model.pthload checkpoint from local path: /media/dsp-slam/weights/pointpillars/model.pth[pydebug] KITIISequence init done	/home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1634272168290/work/aten/src/ATen/native/TensorShape.cpp:2157.)  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]3D detector takes 0.669318 seconds/home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/mmdet/datasets/utils.py:68: UserWarning: "ImageToTensor" pipeline is replaced by "DefaultFormatBundle" for batch inference. It is recommended to manually replace it in the test data pipeline in your config file.  'data pipeline in your config file.', UserWarning)/home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/mmdet/core/anchor/anchor_generator.py:323: UserWarning: ``grid_anchors`` would be deprecated soon. Please use ``grid_priors``   warnings.warn('``grid_anchors`` would be deprecated soon. '/home/dlr/3rdpack/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/mmdet/core/anchor/anchor_generator.py:360: UserWarning: ``single_level_grid_anchors`` would be deprecated soon. Please use ``single_level_grid_priors``   '``single_level_grid_anchors`` would be deprecated soon. '2D detctor takes 0.141035 seconds250 depth samples on the car, 450 rays in totalReconstruction takes 0.238731 seconds```Then I run ./dsp_slam_mono on kitti with online mode:```(dsp-slam) ***@***.***:~/Project/DSP-SLAM$ ./dsp_slam_mono Vocabulary/ORBvoc.bin configs/KITTI04-12.yaml /media/dsp-slam/data/kitti/07 map/kitti/07DSP-SLAM: Object Oriented SLAM with Deep Shape Priors.This program comes with ABSOLUTELY NO WARRANTY;This is free software, and you are welcome to redistribute itunder certain conditions. See LICENSE.txt.Input sensor was set to: MonocularLoading ORB Vocabulary. This could take a while...Vocabulary loaded![debug]: py::module::import(sys)[debug]: py::module::import(reconstruct.utils)[debug]: pyCfgPath: configs/config_kitti.json[debug]: io_utils.attr(get_decoder)(pyCfg)[debug]: strSequencePath: /media/dsp-slam/data/kitti/07[pydebug] to get_seq	[pydebug] to Class KITIISequence	[pydebug] load_calib() done	[pydebug] data_type == KITTI	Segmentation fault (core dumped)
    
        ```
        NOTE the yellow lines are different, the print infos are from ./reconstruct/__init__.py:```def get_detectors(configs):    if configs.detect_online:        from .detector2d import get_detector2d        if configs.data_type == "KITTI":            print("[pydebug] data_type == KITTI\t") # ok            from .detector3d import get_detector3d # <- stuck  only kitti(online detect) will do this line            print("[pydebug] import get_detector3d\t") # no            return get_detector2d(configs), get_detector3d(configs)        else:            return get_detector2d(configs)    else:        if configs.data_type == "KITTI":            return None, None        else:            return None```Apprently, when the program invoke python module in System.cc, it can not go over  from .detector3d import get_detector3d  ,when test other data or kitti with offline, the program won't reach from .detector3d import get_detector3dSo now I duess it is related to communication between c++ and python (pybind11)? Since I am not familiar with pybind11, It would be great to hear your advices  io(￣▽￣)ｄBest,Longrui
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                dlr1432984585
                            
                    
                        
                            
                                    ***@***.***
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

I have tested from my side and cannot reproduce the error. Did you make sure that everything is the same when you run on KITTI and mono sequences? I'm sorry but I have no idea what might be happening. You can reach out to me via ***@***.*** and we can have a more detailed discussion via email.

closed 2021-12-05 by LongruiDong