JingwenWang95/DSP-SLAM #3: Segfault

xinzhichao, 2021-12-14:
Hello, this is a great job. When I run this program, a segfault occurs. I modified the path in config_kitti.json according to the config file in your program, and the environment compiles without any errors. May I ask this What could be the reason?

xinzhichao, 2021-12-14:
But when i run  Freiburg Cars dataset ,is no problems.

xinzhichao, 2021-12-14:
But when i run Freiburg Cars dataset using dsp_slam_mono,is no problems.

JingwenWang95, 2021-12-14:
Hi! Thanks for your interest! It's a bit hard to tell what's happenning as multiple reasons might cause Segfault. Can you please attach your error message?

But given your description, I suspect it might be caused by the installation of mmdetection3d, as only KITTI sequence use 3D detector. Have you run the building script when buiilding DSP-SLAM or have you followed all the steps as in the buidling script?

JingwenWang95, 2021-12-20:
I will close this as there is no updates for a long time.

closed 2021-12-20 by JingwenWang95