JingwenWang95/DSP-SLAM #4: The experiment directory does not include specifications file "specs.json"

Leonard-Yao, 2021-12-16:
Hi,
I can't wait to follow this great job. After successfully built DSP-SLAM , I occur a weird problem.

> **DSP-SLAM: Object Oriented SLAM with Deep Shape Priors.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.
> Input sensor was set to: Stereo
> Loading ORB Vocabulary. This could take a while...
> Vocabulary loaded!
> terminate` called after throwing an instance of 'pybind11::error_already_set'
  what():  Exception: The experiment directory does not include specifications file "specs.json"**

Please tell me what's going wrong, thanks in advance.

JingwenWang95, 2021-12-16:
Hi,
This error suggests that you provided wrong path for DeepSDF weights. Did you download all the weights and put them under `weights/` folder? The json config files under `configs/` assume that all the weights are under `weights/`.

Leonard-Yao, 2021-12-17:
@JingwenWang95 
Thank you for your reply. The problem has been solved as you said.
> This error suggests that you provided wrong path for DeepSDF weights. Did you download all the weights and put them under weights/ folder? The json config files under configs/ assume that all the weights are under weights/.

By the way numpy version 1.19.5 may lead to some error as follows:

> ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject

when this happened, numpy version need to update.

JingwenWang95, 2021-12-17:
@Leonard-Yao Thanks a lot! I did spot the numpy issue before but I forgot to mention that in readme. I will put that in FAQ section.
I will close this. But feel free to reopenit in case you have othere issues.

closed 2021-12-17 by JingwenWang95