JingwenWang95/DSP-SLAM #7: munmap_chunk(): invalid pointer

Taekbum, 2021-12-22:
Hi, thank you for this nice project.
I successfully build dsp_slam with cuda 11.1, pytorch 1.8.2 lts referring to build_cuda113.sh
BUT "munmap_chunk(): invalid pointer" error occurred when I tried to run both dsp_slam and dsp_slam_mono.
Do you have any ideas about this?

JingwenWang95, 2021-12-22:
Hi @Taekbum, Thanks for your interests!

Actually I've ever met a very similar problem before when I developed this project. May I ask you for more details to confirm:
1. Did you make sure your Anaconda cudatoolkit version matches your system CUDA under /usr/local/?
2. Can you successfully run reconstruct_frame.py?

I think I met this problem when I was using CUDA 11.1 and PyTorch 1.8.1. I can run the Python script but met exactly the same error when embedding the Python script into C++. I Googled it for a long time but did not find any solutions. In the end I solved this simply by re-installing PyTorch 1.8.0. I've got the feeling that similar things happen with PyTorch 1.8.2

I assume your nvidia driver does not support CUDA 11.3 and you don't want to upgrade it at the moment, and that's why you didn't run the buiding script? So I would suggest you try reinstalling PyTorch 1.8.0 first and see if that solves that error. Feel free to contact me if you have any further questions.

Taekbum, 2021-12-23:
1. My anaconda cudatoolkit version and system CUDA version were both 11.1 & 2. I successfully run reconstruct_frame.py.
As you recommend, I installed CUDA 11.3 & pytorch 1.10 and succesfully run dsp_slam! 
Thanks for your help.

JingwenWang95, 2021-12-23:
Nice! Feel free to reopen this if you meet any new problems!

closed 2021-12-23 by JingwenWang95