JingwenWang95/DSP-SLAM #8: There is no response after I entering the command~

QingWind6, 2021-12-25:
Hi,I don't konw why there is no response after typing the command.
Here's the command:./dsp_slam_mono Vocabulary/ORBvoc.bin configs/freiburg_001.yaml freiburg_static_cars_52/car001 map/freiburg/001
The dataset i download is the resulting data which is 4.2GB
Could you help me,please

QingWind6, 2021-12-25:
I've solved the first problem,but now i make another:
DSP-SLAM: Object Oriented SLAM with Deep Shape Priors.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: initfsencoding: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

How can I solve this problem?Thank you~

JingwenWang95, 2021-12-25:
Hi @QingWind6, please refer to this issue #4

QingWind6, 2021-12-28:
Thanks！I have fixed all the issues and run it successfully!

closed 2021-12-28 by JingwenWang95