JingwenWang95/DSP-SLAM #9: can not run ./dsp_slam_mono on kitti dataset

jzx-gooner, 2022-01-02:
anazing work!  
I run 
```
./dsp_slam_mono Vocabulary/ORBvoc.bin configs/KITTI00-02.yaml data/kitti/07 map/kitti/07
```
and it crashed 
```
Start processing sequence ...
Images in the sequence: 1101

New Map created with 165 points
New Keyframe
/home/jzx/miniconda3/envs/dsp-slam/lib/python3.7/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1634272168290/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
3D detector takes 0.880130 seconds
/home/jzx/miniconda3/envs/dsp-slam/lib/python3.7/site-packages/mmdet/datasets/utils.py:69: UserWarning: "ImageToTensor" pipeline is replaced by "DefaultFormatBundle" for batch inference. It is recommended to manually replace it in the test data pipeline in your config file.
  'data pipeline in your config file.', UserWarning)
2D detctor takes 0.838410 seconds
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  KeyError: 'background_rays'

At:
  /home/jzx/DSP-SLAM/reconstruct/utils.py(84): __missing__
  /home/jzx/miniconda3/envs/dsp-slam/lib/python3.7/site-packages/addict/addict.py(67): __getattr__

[1]    17664 abort (core dumped)  ./dsp_slam_mono Vocabulary/ORBvoc.bin configs/KITTI00-02.yaml data/kitti/07 
```

JingwenWang95, 2022-01-02:
Hi @jzx-gooner, in our released version monocular input is not supported for kitti sequence. Have you tried running `dsp_slam` insdead? 
Also I noticed in your command you are running on kitti-07 but you are supplying the config file for 00-02. This will not cause crash but will of course affect the performance.

jzx-gooner, 2022-01-03:
> Hi @jzx-gooner, in our released version monocular input is not supported for kitti sequence. Have you tried running `dsp_slam` insdead? Also I noticed in your command you are running on kitti-07 but [...]

Ok.i got it ! thank you very much ~~~拜个早年！

JingwenWang95, 2022-01-06:
新年快乐！

closed 2022-01-06 by JingwenWang95