JingwenWang95/DSP-SLAM #11: How can i use stereo only  mode for kitti ?

Fitz-liu, 2022-01-06:
Thanks for yours great job!

Recently, I'm learning about your project. How to use stereo only mode for Kitti dataset?

JingwenWang95, 2022-01-06:
Hi @Fitz-liu, unfortunately the current released version doesn't support stereo-only mode for KITTI. Actually we've implemented stereo-only version for KITTI but the result is not very good, that's why we didn't open source that version. But it should be fairly simple to develop the stereo-only version on your own by referring to the monocular code.

Fitz-liu, 2022-01-06:
Thanks for your answer! I will try it... By the way , do you have any plans to upgrade this project  from orb-slam2 to orb-slam3?

JingwenWang95, 2022-01-06:
Actually that's a good idea! I will try to migrate my code to ORB-SLAM3 in my spare time, and any contribution will be welcome!

closed 2022-01-08 by JingwenWang95