JingwenWang95/DSP-SLAM #12: When run dsp_slam_mono, it happens "No module named 'skimage'"

HJMGARMIN, 2022-01-06:
**Amazing work!

I run dsp_slam_mono,but I meet the problem,I install skimage by "sudo apt-get install python-skimage",but it still exists.**

(dsp-slam) ubuntu-slam@ubuntuslam-B560M-AORUS-ELITE:~/DSP-SLAM$ ./dsp_slam Vocabulary/ORBvoc.bin configs/KITTI04-12.yaml /home/ubuntu-slam/dataset/KITTI/06 map/kitti/06

DSP-SLAM: Object Oriented SLAM with Deep Shape Priors.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  ModuleNotFoundError: No module named 'skimage'

At:
  /home/ubuntu-slam/DSP-SLAM/reconstruct/utils.py(23): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(728): exec_module
  <frozen importlib._bootstrap>(677): _load_unlocked
  <frozen importlib._bootstrap>(967): _find_and_load_unlocked
  <frozen importlib._bootstrap>(983): _find_and_load

已放弃 (核心已转储)

JingwenWang95, 2022-01-06:
Hi @HJMGARMIN, it's weird to see that you are installing via `sudo apt-get`. Have you tried with `pip install scikit-image`?

HJMGARMIN, 2022-01-06:
> Hi @HJMGARMIN, it's weird to see that you are installing via `sudo apt-get`. Have you tried with `pip install scikit-image`?

Thanks,
I tried,but I ran into a version problem of scikit-image, could tell me your version of scikit-image?
I used 0.19.1,but its networkx and numpy conflict with mmdet3d.

JingwenWang95, 2022-01-06:
In my environment it was 0.18.3. But how did you install DSP-SLAM? If you run the building script these should have already been set up.

HJMGARMIN, 2022-01-07:
> In my environment it was 0.18.3. But how did you install DSP-SLAM? If you run the building script these should have already been set up.

I run the build_cuda102.sh successfully, but when I run dsp_slam, some problems happen.

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  ModuleNotFoundError: No module named 'skimage'

At:
  /home/ubuntu-slam/DSP-SLAM/reconstruct/utils.py(23): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(728): exec_module
  <frozen importlib._bootstrap>(677): _load_unlocked
  <frozen importlib._bootstrap>(967): _find_and_load_unlocked
  <frozen importlib._bootstrap>(983): _find_and_load

已放弃 (核心已转储)

JingwenWang95, 2022-01-07:
> > In my environment it was 0.18.3. But how did you install DSP-SLAM? If you run the building script these should have already been set up.
> 
> I run the build_cuda102.sh successfully, but when I run dsp_slam, some problems happen.
> 
> Loading ORB Vocabulary. This could take a while... Vocabulary loaded!
> 
> terminate called after throwing an instance of 'pybind11::error_already_set' what(): ModuleNotFoundError: No module named 'skimage'
> 
> At: /home/ubuntu-slam/DSP-SLAM/reconstruct/utils.py(23): (219): _call_with_frames_removed (728): exec_module (677): _load_unlocked (967): _find_and_load_unlocked (983): _find_and_load
> 
> 已放弃 (核心已转储)

So you still have trouble installing skimage? Have you tried pip install version 0.18.3?

HJMGARMIN, 2022-01-08:
> So you still have trouble installing skimage? Have you tried pip install version 0.18.3?

Thanks for your reply, I have solved it by "conda install scikit-image", but I have other problems, such as I run dsp_slam, it tels me " Keyerror:'Tr' ".I used your KITTI dataset 07.
What's more, I run dsp_slam_mono on redwood, it tells me ModuleNotFoundError: No module named 'mmcv', however, I intalled mmcv-full successfully.

closed 2022-06-26 by JingwenWang95