JingwenWang95/DSP-SLAM #14: why can not reconstruct object?

mundanePeo, 2022-01-06:
After I construct env and complie project , I run "./dsp_slam Vocabulary/ORBvoc.bin configs/KITTI04-12.yaml data/kitti/07 map/kitti/07" and get some files ,then run "python extract_map_objects.py --config configs/config_kitti.json --map_dir map/07 --voxels_dim 64" get some npy/ply files. But when I run "python visualize_map.py --config configs/config_kitti.json --map_dir map/07", I get lots of black point in open3d.
How can I do to generate the picture you put int the readme "Save and visualize map"?
And the same as freiburg/001.
By the way How can I use ./dsp_slam to run freiburg/001?
tks so much!

JingwenWang95, 2022-01-06:
Hi @mundanePeo, I habe no idea what hapened given you explanation. Can you please upload a screenshot of what you get from Open3D? To run on freiburg sequence you need to run ./dsp_slam_mono, please refer to readme for more details.

JingwenWang95, 2022-01-06:
Also, can you open and visualize the generated ply files using meshlab?

JingwenWang95, 2022-01-07:
Hi @mundanePeo, so this means that the object meshes are already reconstructed but they didn't show up in Open3D. Did you make sure you provided correct data path to the visualization script? I saw when you run DSP-SLAM you saved the map to `map/kitti/07` whereas when you run the visualization script you are using `map/07`. Can you make them consistent and try again?

JingwenWang95, 2022-01-07:
Or you could simply look into the visualization script and change the code to make sure the object meshes are correctly loaded. What the visualization script does is simply to load all the meshes and points and show them in Open3D. Given that all the object meshes and points are already there, what you need to do is just to make sure that they are correctly loaded. So I'd recommend you to dive into the visualization script and figure out what happened.

mundanePeo, 2022-01-08:
> 

Firstly, tks for your reply. 
I make sure the path is correct. becuase I used path "data/kitti/07" as data path and path "map/kitti/07" as map_dir when I run the ./dsp_slam. Before I run the visualization script, I copied MapObjects.txt and MapPoints.txt from "map/kitti/07" to "map/07".

mundanePeo, 2022-01-08:
> Hi @mundanePeo, so this means that the object meshes are already reconstructed but they didn't show up in Open3D. Did you make sure you provided correct data path to the visualization script? I saw [...]

And when I first run the visualization script, the map_dir is "map/kitti/07". And the result is same.

JingwenWang95, 2022-01-08:
> > @JingwenWang95
> 
> Firstly, tks for your reply. I make sure the path is correct. becuase I used path "data/kitti/07" as data path and path "map/kitti/07" as map_dir when I run the ./dsp_slam. Before I run the visualization script, I copied MapObjects.txt and MapPoints.txt from "map/kitti/07" to "map/07".

If all the paths are correct, the visualization script should load the meshes and show them on the screen. Also I notioced your mesh is stored at `map/07/object/0.ply`. This shouldn't happen because `extract_map_objects.py` will put all the meshes under `map_dir/objects/`. And `visualize_map.py` will also search under the default subdirectory. But I don't know what happened yours are all stored under `object`. That's why I recommend you to look into `visualize_map.py`, change the corresponding lines of code and load the meshes yourself.

mundanePeo, 2022-01-10:
> If all the paths are correct, the visualization script should load the meshes and show them on the screen. Also I notioced your mesh is stored at `map/07/object/0.ply`. This shouldn't happen because [...]

ok，tks! I get it！

closed 2022-01-14 by JingwenWang95