JingwenWang95/DSP-SLAM #15: mmdet3d error

HJMGARMIN, 2022-01-06:
Hello, I run into a problem about mmdetection3d, I don't how to solve it, I have confused for hours.

`Traceback (most recent call last):
  File "setup.py", line 248, in <module>
    zip_safe=False)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/setuptools/command/develop.py", line 114, in install_for_development
    self.run_command('build_ext')
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 735, in build_extensions
    build_ext.build_extensions(self)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/command/build_ext.py", line 534, in build_extension
    depends=ext.depends)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 483, in unix_wrap_single_compile
    cflags = unix_cuda_flags(cflags)
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 450, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/home/ubuntu-slam/anaconda3/envs/dsp-slam/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1606, in _get_cuda_arch_flags
    arch_list[-1] += '+PTX'
IndexError: list index out of range
`

JingwenWang95, 2022-01-06:
I'm sorry but I have no idea what happened given only those error messgaes. Could you please provide more details on how you did the installation? such as: Did you install DSP-SLAM by running the building script? If not, did you follow all the steps in the building script and readme? Did you make sure your system CUDA version matches that of Anoconda cudatoolkit? Was all the installation successful? Was this error happened when you run the C++ code or python-only code?

HJMGARMIN, 2022-01-06:
> I'm sorry but I have no idea what happened given only those error messgaes. Could you please provide more details on how you did the installation? such as: Did you install DSP-SLAM by running the [...]

OK, I installed DSP-SLAM by running the building script build_cuda102.sh. The problem happened at "pip install -v -e" after "cd mmdetection3d" when I run the build script.
I think if there is a possibility that the mmdetection3d version has been updated after you uploaded the project, so I failed.

HJMGARMIN, 2022-01-06:
> I'm sorry but I have no idea what happened given only those error messgaes. Could you please provide more details on how you did the installation? such as: Did you install DSP-SLAM by running the [...]

In addition to mmdetection3d, the other dependencies all installed successfully.

JingwenWang95, 2022-01-06:
So you installed everything using the building script including the system CUDA? It's weird as the error message suggests that CUDA archetecture flag is not set. So It's very likely that your CUDA or PyTorch installation was not successful. Have you checked if `torch.cuda.is_available()` is True or not?

HJMGARMIN, 2022-01-06:
> So you installed everything using the building script including the system CUDA? It's weird as the error message suggests that CUDA archetecture flag is not set. So It's very likely that your CUDA [...]

Oh!Thank you! I found I got a wrong version of cuDNN.

closed 2022-01-08 by JingwenWang95