JingwenWang95/DSP-SLAM #19: Stereo-only input support?

qinyq, 2022-04-07:
Hi, thanks for your nice work and sharing the code! I was quite happy when I see the "stereo streams" are supported in the paper. I could successfully reproduce the result (qualitatively). 

However, it seems that the released code does not support stereo-only input, right? The kitti dataset uses stereo and lidar data at the same time. Do I miss anything or if there do have support? Thank you!

JingwenWang95, 2022-04-07:
Hi, thanks for your interest! Unfortunately we didn't release the stereo-only code because the result was not so good. You can also refer to #11

qinyq, 2022-04-08:
> Hi, thanks for your interest! Unfortunately we didn't release the stereo-only code because the result was not so good. You can also refer to #11

Thanks! It's a pity to hear that the stereo-only not working well. Have you got any idea of why? Is it because finding corresponding points? or triangulation?

JingwenWang95, 2022-04-10:
> Thanks! It's a pity to hear that the stereo-only not working well. Have you got any idea of why? Is it because finding corresponding points? or triangulation?

Actually there are several reasons, but we didn't have enough time to conduct more thourough analysis to try to solve this. But the two major issues I observed were: 1. the initial pose estimate is way worse. 2. the SLAM points are way noisier. I'm happy to have further discussions and you can reach out to me via email.

closed 2022-08-17 by qinyq