JingwenWang95/DSP-SLAM #22: sv2_cars_filtered.json

alanxuefei, 2022-06-20:
Would you like to share the sv2_cars_filtered.json file?
Because I try to extend DSP-SLAM into more objects. sv2_cars_filtered.json could be an example. 

```
  "Description" : [ "This experiment learns a shape representation for cars ",
                    "using data from ShapeNet version 2." ],
  "DataSource" : "/media/jingwen/Data/deepsdfData",
  "TrainSplit" : "/home/jingwen/Vision/dev/slam_with_shape_priors/DeepSDF/examples/splits/sv2_cars_filtered.json",
  "TestSplit" : "/home/jingwen/Vision/dev/slam_with_shape_priors/DeepSDF/examples/splits/sv2_cars_filtered.json",
```

JingwenWang95, 2022-06-20:
Hi @alanxuefei, Sorry I have done the training very long time ago and reformatted my hard drive in the mean time, so I don't have the split file any more. But they follow the same format as the example split files provided in DeepSDF repository: https://github.com/facebookresearch/DeepSDF/tree/main/examples/splits
which basically contains the instance names in ShapeNet. You can create your own file by following the same format.

nnop, 2023-01-09:
Hi, I didn't find this file in the repo. Which commit contains this file?