JingwenWang95/DSP-SLAM #23: undefined reference to `TIFF****@LIBTIFF_4.0'

Hello-Water, 2022-08-16:
When you meet the following problem,
"
../Thirdparty/Pangolin/build/libpango_image.so: undefined reference to `TIFFOpen@LIBTIFF_4.0'
../Thirdparty/Pangolin/build/libpango_image.so: undefined reference to `TIFFGetField@LIBTIFF_4.0'
../Thirdparty/Pangolin/build/libpango_image.so: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
../Thirdparty/Pangolin/build/libpango_image.so: undefined reference to `TIFFReadScanline@LIBTIFF_4.0'
../Thirdparty/Pangolin/build/libpango_image.so: undefined reference to `TIFFClose@LIBTIFF_4.0'
"
add "-ltiff" to "targtet_link_libraries" in "DSP-SLAM/CMakeLists.txt". It may be helpful.

GetOverMassif, 2022-10-08:
check if you choose the right path of opencv when compiling pangolin?

closed 2023-01-09 by JingwenWang95

vijendra1125, 2024-02-29:
I  built DSP-SLAM using build_cuda113.sh and the suggestion in this issue description helped me building it successfully whereas when I ran the dsp_slam with argument to save the map I got the error `/usr/local/lib/libtiff.so.5: no version information available`. 

I would like to share the solution here in case someone else faces same problem. To resolve this I  created symbolic link for libtiff.so.5  from `/usr/lib/x86_64-linux-gnu` to  `/usr/local/lib`
To do that first either remove or rename the `/usr/local/lib/libtiff.so.5` and then run 
`sudo ln -s /usr/lib/x86_64-linux-gnu/libtiff.so.5 /usr/local/lib/libtiff.so.5`

TINY-KE, 2024-03-15:
I tried all the methods mentioned above, but none of them worked.
I resolved the issue by adding the absolute path of libtiff.so.5 to the **target_link_libraries()**  in CMakeLists.txt. 

Obtained the absolute path of libtiff.so.5 by running the command 
`ldd Thirdparty/Pangolin/build/libpango_image.so | grep tiff`
Then
```
target_link_libraries(dsp_slam ${PROJECT_NAME} /usr/lib/x86_64-linux-gnu/libtiff.so.5)
target_link_libraries(dsp_slam_mono ${PROJECT_NAME} /usr/lib/x86_64-linux-gnu/libtiff.so.5)
```