JingwenWang95/DSP-SLAM #25: run dsp_slam_mono but nothing happens

GetOverMassif, 2022-09-23:
Hello, I have configed the environment and compile the DSP-SLAM project successfully, then I run the dsp_slam_mono as follows. The programme keeps still, without any information、error inform or exit. Could you suggest any probable causes?
`(dsp-slam2) lj@ljc:~/Documents/DSP-SLAM$ ./dsp_slam_mono Vocabulary/ORBvoc.bin configs/redwood_09374.yaml /media/lj/TOSHIBA/dataset/RedwoodOS/09374 map/redwood/09374`

JingwenWang95, 2022-09-24:
Hi @GetOverMassif, the most common cause is you probably provided some paths that don't exist. This issue is inhereited from ORB-SLAM2, the program will have no response if it fails to recognoze some paths. So I would suggest you double check the paths you provided.

GetOverMassif, 2022-09-26:
> 

I found it's exactly the problem of file path, leading to error when reads images. But some new problem occur, I solved some but stuffed here:
`terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  ImportError: cannot import name 'config_decoder' from 'deep_sdf.workspace' (/home/lj/anaconda3/envs/dsp-slam2/lib/python3.7/site-packages/deep_sdf/workspace.py)`
I compiled DeepSDF successfully and copied 'deep_sdf' package into conda env, it seems there is no file called config_decoder.

JingwenWang95, 2022-09-26:
Hi @GetOverMassif You don't need to compile DeepSDF yourself, it's already included [here](https://github.com/JingwenWang95/DSP-SLAM/blob/master/deep_sdf) in this project. And actually this might cause problems as I have made massive changes to the original DeepSDF. You can try removing the DeepSDF you installed from your conda environment and see what happens.

GetOverMassif, 2022-09-27:
> Hi @GetOverMassif You don't need to compile DeepSDF yourself, it's already included [here](https://github.com/JingwenWang95/DSP-SLAM/blob/master/deep_sdf) in this project. And actually this might [...]

Thanks a lot for replying patiently, it seems I can nealy run the dsp_slam. Final question: where can I download the weights as you answer in other issues (to be specific, the label_2D/3D, model.pth and deep_sdf/XXX_64)?

JingwenWang95, 2022-09-29:
Please have a look at here: https://github.com/JingwenWang95/DSP-SLAM#dataset

GetOverMassif, 2022-10-08:
> Please have a look at here: https://github.com/JingwenWang95/DSP-SLAM#dataset

I run your DSP-SLAM successfully finally. Thanks a lot for your patient reply. I list some info that may cause conflict, wishing to help others:
1. Use cuda 10.2 or 11.3 as DSP-SLAM suggests, as I used cuda 11.0 at first and met some strange problems.
2. Add option '-D BUILD_TIFF=ON' when compiling OpenCV to avoid error like 'undefined reference to `TIFFReadDirectory@LIBTIFF_4.0''.
3. The 'environment_cuda113.yml' install 'opencv-python==4.5.4.58' but version of opencv in Thirdparty is 3.4, I uninstalled opencv-python==4.5.4.58 and installed according version of opencv 3.4, which make the system worked.

closed 2022-10-08 by GetOverMassif