JingwenWang95/DSP-SLAM #26: Aborted (core dumped)

shuzhangshu, 2022-09-27:
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  AttributeError: module 'skimage.measure' has no attribute 'marching_cubes_lewiner'

At:
  /home/shu/DSP-SLAM/reconstruct/utils.py(130): convert_sdf_voxels_to_mesh
  /home/shu/DSP-SLAM/reconstruct/optimizer.py(218): extract_mesh_from_code
Aborted (core dumped)
 
Hello author, the above is the error I prompted when running the code. When running the code, the interface will pop up for 5s first, and then it will end, sometimes even shut down.Could you please give me some guidance.

JingwenWang95, 2022-09-29:
Hi @shuzhangshu! The error message is quite clear: it failed to import marching cubes function from skimage.measure on Python side. Marching cubes is only used for extracting meshes of reconstructed objects for visualisation. That's also why this error was not prompted from the beginning when no objects were reconstructed. 

This can sometimes happen due to the version mismatch of skimage. Maybe you can try re-install the exact version of skimage as I provided in the environment file.

shuzhangshu, 2022-10-08:
Thank you very much for your answer! I'm sorry to bother you because I was too careless. I have another question for your guidance. Of the three given running dataset codes, I can only run related KITTI datasets. The other two will be shut down suddenly when running, but KITTI will show some results and then shut down. Could you please ask if this is due to my computer hardware (or video memory. etc)?

shuzhangshu, 2022-10-08:
> Hi @shuzhangshu! The error message is quite clear: it failed to import marching cubes function from skimage.measure on Python side. Marching cubes is only used for extracting meshes of reconstructed [...]
Thank you very much for your answer! I'm sorry to bother you because I was too careless. I have another question for your guidance. Of the three given running dataset codes, I can only run related KITTI datasets. The other two will be shut down suddenly when running, but KITTI will show some results and then shut down. Could you please ask if this is due to my computer hardware (or video memory. etc)?

closed 2023-01-09 by JingwenWang95