JingwenWang95/DSP-SLAM #31: make  build problem

CoderDoubleTen, 2023-04-19:
[ 94%] Linking CXX executable /home/sjm/ql/projects/DSP-SLAM/dsp_slam
/usr/local/lib/libpango_image.so：对‘TIFFOpen@LIBTIFF_4.0’未定义的引用
/usr/local/lib/libpango_image.so：对‘TIFFGetField@LIBTIFF_4.0’未定义的引用
/usr/local/lib/libpango_image.so：对‘TIFFScanlineSize@LIBTIFF_4.0’未定义的引用
/usr/local/lib/libpango_image.so：对‘TIFFSetWarningHandler@LIBTIFF_4.0’未定义的引用
/usr/local/lib/libpango_image.so：对‘TIFFReadScanline@LIBTIFF_4.0’未定义的引用
/usr/local/lib/libpango_image.so：对‘TIFFClose@LIBTIFF_4.0’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/dsp_slam.dir/build.make:134: recipe for target '/home/sjm/ql/projects/DSP-SLAM/dsp_slam' failed
make[2]: *** [/home/sjm/ql/projects/DSP-SLAM/dsp_slam] Error 1
CMakeFiles/Makefile2:129: recipe for target 'CMakeFiles/dsp_slam.dir/all' failed
make[1]: *** [CMakeFiles/dsp_slam.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

molong-d, 2023-05-19:
如果pangolin安装时候报错:pangolin undefined reference to `TIFFScanlineSize@LIBTIFF_4.0’，那么可能是版本问题，我替换成了pangolin0.6[下载链接在这个博主的帖子里](https://blog.csdn.net/qq_44195329/article/details/124197681?spm=1001.2101.3001.6650.3&utm_medium=distribute.pc_relevant.none-task-blog-2~default~CTRLIST~Rate-3-124197681-blog-124184774.pc_relevant_paycolumn_v3&depth_1-utm_source=distribute.pc_relevant.none-task-blog-2~default~CTRLIST~Rate-3-124197681-blog-124184774.pc_relevant_paycolumn_v3&utm_relevant_index=5)，就可以了

Pharaon23, 2023-10-12:
请问你解决了吗？