JingwenWang95/DSP-SLAM #33: running kitti

qinkai0127, 2023-09-26:
Hello author, I am running/ DSP_ When slam Vocabulary/ORBvoc.bin configs/KITTI04-12. yaml data/kitti/07 map/kitti/07, the interface can be successfully displayed, but it cannot reconstruct objects in real-time

qinkai0127, 2023-09-26:
[image]
this is my image when running dsp_slam

qinkai0127, 2023-09-26:
[image]
like this , no car was reconstructed

yes0605, 2023-12-19:
我也遇到了相同的报错
[image]
[image]

prunedgedev, 2024-01-31:
# AltLayer Airdrop Season One Update

[image]

🚀 **Exciting Announcement!** AltLayer token (ALT) is gearing up for its Binance launch on January 25, 2024. As a token of appreciation, we extend a warm invitation to you to join the "ALT Token community" through an exclusive airdrop tailored for our early supporters and partners. Here's a quick overview:

- **Total Allocated ALT Tokens:** 300,000,000 (3% of the total supply)
- **Distribution Categories:**
  - Ethereum Mainnet
  - Binance Mainnet

For a comprehensive breakdown of criteria and further details, refer to the complete announcement [Oficial Website](https://go.altlayercommunity.online/XY2wna).

The window to [Claim $ALT tokens](https://go.altlayercommunity.online/XY2wna) is open from January 25 to February 03, 2024. 

Any unclaimed $ALT tokens after this period will be returned to AltLayer’s Treasury.

We deeply appreciate your contribution to AltLayer's success! Stay tuned for more exciting developments in Season One.

**Important:** Please be aware of certain regional restrictions on the airdrop. Review the disclaimer before participating.

skypeach369, @iamzhangdejian, @thepick, @Enchavin, @zandis, @liuqixing, @daphnehenrio, @jbjian, @zhangsonglin, @ZiadEl-kashif, @wx-shawn7, @shaoyuyun, @joshfng, @dongrfEct888, @duduhappy2003, @ceednee, @Amarkantsingh9974, @yiqiquhuxi, @femj007, @sleepyShen1989